Bastardkb/adapter-blackpill #6: Firmware support for Charybdis (4x6)

0xcharly, 2021-12-19:
Tracking the development of the firmware for the Charybdis (4x6).

- [x] create branch: [0xcharly/qmk_firmware:charybdis-upstream-f411](https://github.com/0xcharly/qmk_firmware/tree/charybdis-upstream-f411)
- [x] create configuration: [bastardkb/charybdis/4x6/f411](https://github.com/0xcharly/qmk_firmware/tree/charybdis-upstream-f411/keyboards/bastardkb/charybdis/4x6/f411)
- [ ] validate configuration
- [ ] validate execution
    - [x] flash board
    - [ ] test keyboard matrix
    - [ ] test RGB
    - [ ] test trackball

bstiq, 2022-02-14:
Because I'm making a new shield, all the firmware needs to be revamped.
The shield should be compatible with the elite-C, pro micro, nice!nano, StemCell, and anyRP2040 based board as long as it's using the same pinout.

With the various improvements brought by the new shield, I will eventually switch all my keyboards to it.
Thanks again to Karl for the work on the blackpill shield, which I used to improve on the elite-c shield.

Here is the WIP on the new shield:
https://github.com/Bastardkb/Elite-C-holder/pull/1

## TODO for each keyboard
### New folder structure

Keyboards will be compatible with 3 shields:

- 1.0 for elite-C only
- blackpill adapter for blackpill only
- 2.0 for elite-C, stemcell, n!n, pro micro

For each keyboard:
1.x
--- u4
2.x
--- stemcell
--- u4
--- pro-micro
--- nice-nano
blackpill
keymaps
--- default
--- stock
--- via

0xcharly, 2022-02-28:
I'm going to clone this.  The blackpill adapter is taking a little bit of a back seat at the moment, but the firmware should already work with the Charybdis 4x6.  There's also new firmware for the Scylla, TBK Mini, and Skeletyl.  So the entire offering is now compatible (firmware-wise) with the Blackpill adapter.

The new folder structure has been implemented, and is tracked by issues in the Elite-C repo.

closed 2022-02-28 by 0xcharly