Bastardkb/adapter-blackpill #9: Adjust position of matrix and SPI pins

0xcharly, 2021-12-30:
Matrix and SPI pins at the bottom of the PCB allow for easy installation/maintenance, but it is not easy, with the current shape of the organic tent, to wire them.  The following points only apply to the PCB assembly on the organic tent (I have not tried yet with the updated blackpill case):

- The ribbon cables are a bit too short to install the blackpill on the organic tent
- The first 2 pins of each columns are obstructed by the organic tent support pillar, further complicating the installation
- The opening in the organic tent is pretty narrow, and it's easy to scratch the prints by trying to insert the PCB+blackpill assembly, in particular when attached to the ribbon cables as they are pretty stiff
- The RGB cables need to be routed on the other side of the support pillar, which complicates maintenance as it is now impossible to remove the assembly without desoldering either the RGB or the matrix/spi wires
- The RGB wires need to be longer to be routed against the case's wall (which otherwise complicates maintenance on some switches because the RGB cables are crossing the plate through the center)

One option could be to angle the matrix/spi pins column by 90 degrees (ie. using rows of pins instead of columns at the bottom of the PCB).  Ribbon cables can easily be installed and bent 90 degrees as to fold against the blackpill, and be routed around the support pillar (north side) along with the RGB cables: all cables would come out through the same side of the pillar.

Note that the pillar position is not the same on the Skeletyl and Charybdis side, so what works better for one side will make it harder for the other side.  It would be ideal to update the tent design to have the pillar at the same position on both halves.

bstiq, 2022-01-20:
Fixed this issue with the new tents which don't have the middle pillar. 
I still have to print them to test strenght, but should solve it.
Also tested the lenght of the ribon cables and it's fine.

Once I have the time to test the tents, will update here and close the PR.

bstiq, 2022-02-14:
Tents tested and confirmed working.

closed 2022-02-14 by bstiq