Bastardkb/adapter-blackpill #11: v1.2

0xcharly, 2022-01-19:
@bstiq @KarlK90 

List of things for the `v1.2`.  I'm including first the things I believe we have already agreed upon, please keep me honest:

- Move the `UPDATE` button 0.3mm up (please double check these measurements), so that it's equally accessible through the base plate on both the Skeletyl and Charybdis 3x5 sides
  - TODO(@bstiq) confirm that hex pattern placement is the same for Scylla/Mini/Charybdis 4x6
- Reduce the size of the PCB around the top screw holder, to avoid squeezing the wires under it (the matrix wires coming out of the case PCB)

## Other suggestions

These suggestions need further discussion:

- Swap matrix and SPI pinout so that matrix pinout is on the outer side
  - The goal is to make more room for the ribbon/wires used for the matrix: currently, the adapter pinouts are almost aligned with the case PCB pinout, which puts a lot of stress on both pinouts and wires
- Rotate pinout on the adapter by 90 deg
  - This is an alternative to reduce the stress on the wires/ribbons

KarlK90, 2022-01-19:
Also adding another column of pins for I2C to the left or right of the matrix and SPI pins.

0xcharly, 2022-01-19:
Added.

bstiq, 2022-01-19:
> Reduce the size of the PCB around the top screw holder, to avoid squeezing the wires under it (the matrix wires coming out of the case PCB)

As long as the screw placement remains the same, you can go ahead
[adapter-Edge_Cuts.zip](https://github.com/Bastardkb/adapter-blackpill/files/7897443/adapter-Edge_Cuts.zip)
Eg this one will work

bstiq, 2022-01-19:
Btw didn't really check - is the row and col placement on this one the same as the original ?

bstiq, 2022-01-19:
@0xcharly 

> TODO(@bstiq) confirm that hex pattern placement is the same for Scylla/Mini/Charybdis 4x6

I confirm for TBK Mini, skeletyl 
For the charybdis nano it's slightly different.
Button should be moved up 3mm

0xcharly, 2022-01-19:
@bstiq Got it, thanks.  Is this the same for Charybdis 3x5 vs. 4x6, and for Skeletyl/TBK Mini vs Scylla?

Looking at the pictures on Discord, the hole is 1cm wide, and the center of the button is at 0mm on the Charybdis side, and 2.5mm on the Skeletyl side.  Thus, moving the button 3.75mm up should make it symmetrical: respectively 3.75mm and 6.25mm, which puts both of them 1.25mm away from the center of the hex hole.  Does that make sense for you?

0xcharly, 2022-01-19:
How about moving the blackpill footprint 0.5mm away from the TRS connector?  That way it you actually align with the cutouts on the blackpill cases?  (instead of adjusting the case.)

bstiq, 2022-01-20:
I can modify the plates of the Scylla and Chary 4x6 easily, the priority is skeletyl/mini/CNano.

I moved the blackpill 0.5mm closer to the TRS connector because there were overlapping issues with the screw on the left. There was not enough space for the 3d printed part that holds the screw insert.
Because of that I'm stuck with making new versions of the cases.

bstiq, 2022-01-20:
@KarlK90 so I didn't quite get the changes to the trackball connector, is the pinout still compatible with my sensor PCB?

0xcharly, 2022-01-20:
> I can modify the plates of the Scylla and Chary 4x6 easily, the priority is skeletyl/mini/CNano.
>
> I moved the blackpill 0.5mm closer to the TRS connector because there were overlapping issues with [...]

Got it.

bstiq, 2022-01-20:
With that additional I2C connector, could we connect eg a trackpad or a rotary encoder ?

bstiq, 2022-01-20:
Also this is switched to TRSS from TRS right ? nice to have a trackball or others on the slave part.
But is there a way to make it TRS-optional, based on firmware? For those who don't need an input device on slave side and/or want to keep using TRS cables rather than TRSS (easier to find)

KarlK90, 2022-01-20:
Started PR #12 for revision 1.2